vita-epfl/CrowdNav #5: how to train/test cadrl and lstm_rl?

CxyCH, 2018-11-17:
Hello:
     First of all, thank you for your open source code. I want to know how to train/test the cadrl and lstm_rl model? when I try to use this：
 `python test.py --policy cadrl --model_dir data/output --phase test` or add `--il` like this:
`python test.py --policy cadrl --model_dir data/output --phase test --il`
  I get this error:
`2018-11-17 21:58:04, INFO: Using device: cpu
2018-11-17 21:58:04, INFO: Policy: CADRL without occupancy map
Traceback (most recent call last):
  File "test.py", line 123, in <module>
    main()
  File "test.py", line 60, in main
    policy.get_model().load_state_dict(torch.load(model_weights))
  File "/home/tcj/.local/lib/python3.5/site-packages/torch/nn/modules/module.py", line 719, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for ValueNetwork:
	Missing key(s) in state_dict: "value_network.0.bias", "value_network.0.weight", "value_network.2.bias", "value_network.2.weight", "value_network.4.bias", "value_network.4.weight", "value_network.6.bias", "value_network.6.weight". 
	Unexpected key(s) in state_dict: "mlp1.0.weight", "mlp1.0.bias", "mlp1.2.weight", "mlp1.2.bias", "mlp2.0.weight", "mlp2.0.bias", "mlp2.2.weight", "mlp2.2.bias", "attention.0.weight", "attention.0.bias", "attention.2.weight", "attention.2.bias", "attention.4.weight", "attention.4.bias", "mlp3.0.weight", "mlp3.0.bias", "mlp3.2.weight", "mlp3.2.bias", "mlp3.4.weight", "mlp3.4.bias", "mlp3.6.weight", "mlp3.6.bias".`

CxyCH, 2018-11-21:
Looks like it was a version problem?

ChanganVR, 2018-11-21:
Hi, the codes have some issue with training cadrl, but your problem is different. Have you trained the model before testing it? Btw, I have fixed the issue I found, could you try the latest code again?
Traning cadrl:
` python train.py --policy cadrl
`
After its done, run:
`
py test.py --policy cadrl --model_dir data/output/ --phase test
`

CxyCH, 2018-11-22:
Thank you, it help for me

closed 2018-11-22 by CxyCH