vita-epfl/CrowdNav #9: Questions about action sampling

LeeKeyu, 2019-03-18:
Hi, I have some questions about the following code in "build_action_space" function (in cadrl.py,line 82):
**speeds = [(np.exp((i + 1) / self.speed_samples) - 1) / (np.e - 1) * v_pref for i in range(self.speed_samples)]**
The comment said the actions were sampled **uniformly**. Hence I cannot understand why the speeds are given by exponential function within (0,1]?
Thank you very much!

ChanganVR, 2019-03-18:
We experimented with both uniform and exponential sampling, the exponential gives slightly better performance. The intuitive reason is that exponential sampling yields samples with smaller speeds, which gives the agent more flexibility to control its actions. For example, when the distance from the goal to the robot is very small, it can stop exactly at the goal position rather than pass it due to the discrete action space.

LeeKeyu, 2019-03-19:
Thank you, it makes sense.

closed 2019-03-19 by ChanganVR