vita-epfl/CrowdNav #11: Question about experiment

CxyCH, 2019-04-02:
Hi:
   The segway is a differential drive robot, but your paper assumes the robot is holonomic? So what is the difference between these two kinematic modes in your method?

ChanganVR, 2019-04-03:
In the paper, we experimented on holonomic mode. When we deployed our algorithm on the robot, we found the kinematics is different. So we added another unicycle kinematic mode in the library. So the actual policy we deployed was trained under similar kinematic to the robot. You can switch kinematic mode by setting kinematics = holonomic in policy.config.

closed 2019-04-08 by ChanganVR

JiayunjieJYJ, 2023-06-02:
> In the paper, we experimented on holonomic mode. When we deployed our algorithm on the robot, we found the kinematics is different. So we added another unicycle kinematic mode in the library. So the [...]

How to run a unicycle kinematic mode ORCA in your code? Thanks!

ChanganVR, 2023-06-05:
You can't. ORCA has a fixed holonomic setting. You can change the setting for the robot in the config file though.

JiayunjieJYJ, 2023-06-07:
> @JiayunjieJYJ You can't. ORCA has a fixed holonomic setting. You can change the setting for the robot in the config file though.
What will be the impact of changing the configuration file to unicycle kinematic mode? Will the learning-based navigation method be changed to a unicycle kinematic mode, but ORCA-based baselines will not be affected?

ChanganVR, 2023-06-12:
>What will be the impact of changing the configuration file to unicycle kinematic mode? Will the learning-based navigation method be changed to a unicycle kinematic mode, but ORCA-based baselines will not be affected?

yes, that is true