vita-epfl/CrowdNav #12: Problem of importing crowd_sim

SamNew1, 2019-06-02:
When I run 'python train.py --policy sarl', I got the following problem: 

Traceback (most recent call last):
  File "train.py", line 10, in <module>
    from crowd_sim.envs.utils.robot import Robot
  File "/home/sam/CrowdNav/crowd_sim/envs/__init__.py", line 1, in <module>
    from crowd_sim.envs import crowd_sim
  File "/home/sam/CrowdNav/crowd_sim/envs/crowd_sim.py", line 548
    nonlocal global_step
                       ^
SyntaxError: invalid syntax

Would you please help me to solve this problem? Thank you very much!

ChanganVR, 2019-06-04:
Hi. Which python version do you use? The nonlocal syntax only works in python 3 and I assume you're using python 2 right?

SamNew1, 2019-06-05:
I used python 3.6, but still got that problem

ChanganVR, 2019-06-05:
I have run the code with python 3.6 and it worked just fine. I doubt it might be an issue of your system/environment configuration, e.g. not switching to the correct virtualenv.

closed 2019-06-07 by ChanganVR