vita-epfl/CrowdNav #14: Questions about the policy of human

ZC119, 2019-08-02:
Hi! I want to know could human's policy be CADRL or SARL for the purpose of Crowd Simulation? I noticed that you have implemented ORCA and Linear policy for human. But I don't know how to replace ORCA with CADRL or SARL. Could you give me some advice? Thank you!

ChanganVR, 2019-08-03:
Hi, Thank you for your interest. Do you mean you want to replace the policy with CADRL or SARL in training or test? If you're going to test them in a multiagent setting, it should be quite straightforward. You probably need to change the settings in the configuration and load the trained model for all policies. But for training, our repo currently doesn't support that. There would be a quite significant change to switch to multiagent training.

ZC119, 2019-08-03:
Thank you for your reply! I will test them first.

closed 2019-08-03 by ChanganVR