vita-epfl/CrowdNav #15: train.py error

liangjiaxuan1018, 2019-08-27:
When I  train a policy,the error as follows：
Output directory already exists! Overwrite the folder? (y/n)n
Traceback (most recent call last):
  File "train.py", line 177, in <module>
    main()
  File "train.py", line 57, in main
    repo = git.Repo(search_parent_directories=True)
  File "/home/ljx/anaconda3/envs/CrowdNav/lib/python3.6/site-packages/git/repo/base.py", line 184, in __init__
    raise InvalidGitRepositoryError(epath)
git.exc.InvalidGitRepositoryError: /home/ljx/桌面/CrowdNav-master/crowd_nav

Is it a problem with the installation package version?

liangjiaxuan1018, 2019-08-28:
Thank you for your reply.I tried to re-run this program, but I still encountered the following problems:

(CrowdNav) ljx@ljx-ChengMing-3967:~/CrowdNav-master/crowd_nav$ python train.py --policy sarl
Output directory already exists! Overwrite the folder? (y/n)y
Traceback (most recent call last):
  File "train.py", line 179, in <module>
    main()
  File "train.py", line 59, in main
    repo = git.Repo(search_parent_directories=True)
  File "/home/ljx/anaconda3/envs/CrowdNav/lib/python3.6/site-packages/git/repo/base.py", line 184, in __init__
    raise InvalidGitRepositoryError(epath)
git.exc.InvalidGitRepositoryError: /home/ljx/CrowdNav-master/crowd_nav

Is it a problem with the git package?

ChanganVR, 2019-08-29:
It looks like something is wrong with the git module. What I can find online is this link: https://github.com/gitpython-developers/GitPython/issues/255. Try it out and see if this error can be fixed. 
In the worse case, you can just comment out these lines of codes, which use the git module. The git module is mainly used for keeping track of code versions when doing development.

closed 2019-09-02 by ChanganVR

luxiaojuntokyo, 2019-12-04:
I got the same error.
(venv) luxiaojun@luxiaojunP50:~/simulationbase/CrowdNav-master/crowd_nav$ python train.py --policy sarl
Output directory already exists! Overwrite the folder? (y/n)y
Traceback (most recent call last):
  File "train.py", line 177, in <module>
    main()
  File "train.py", line 57, in main
    repo = git.Repo(search_parent_directories=True)
  File "/home/luxiaojun/simulationbase/venv/lib/python3.6/site-packages/git/repo/base.py", line 184, in __init__
    raise InvalidGitRepositoryError(epath)
git.exc.InvalidGitRepositoryError: /home/luxiaojun/simulationbase/CrowdNav-master/crowd_nav

Is there any result in this problem?
Thank you.

ChanganVR, 2019-12-05:
Did you run the code outside a git repo?

luxiaojuntokyo, 2019-12-05:
Changan
Thank you for your reply.
When I run "git status" under this directory (~/simulationbase/CrowdNav-master/crowd_nav), “fatal: Not a git repository (or any of the parent directories)” will occur.

Previously, I just download the zip file, extract, build and install.
Now I use git clone and then build and install. The problem didn't apprear.

ChanganVR, 2019-12-05:
That will be the solution to solve this problem. For other people, if you run into this problem as well, make sure you run the code inside the git repo.

asenasen123, 2020-06-04:
> When I train a policy,the error as follows：
> Output directory already exists! Overwrite the folder? (y/n)n
> Traceback (most recent call last):
> File "train.py", line 177, in 
> main()
> File "train.py", line 57, in main
> repo = git.Repo(search_parent_directories=True)
> File "/home/ljx/anaconda3/envs/CrowdNav/lib/python3.6/site-packages/git/repo/base.py", line 184, in **init**
> raise InvalidGitRepositoryError(epath)
> git.exc.InvalidGitRepositoryError: /home/ljx/桌面/CrowdNav-master/crowd_nav
> 
> Is it a problem with the installation package version?

你好！我也是这个研究方向，有些问题想要请教，能否价格好友学习交流下，qq:1414396376

ChanganVR, 2020-06-07:
The git path shouldn't have Chinese characters.

asenasen123, 2020-06-08:
你好！我是这个方向的研究生。我想请问下，如何在这篇论文代码里训练CADRL以及看到可视化结果，我应该怎么处理？我还有些问题想请教你，能否加下你的微信，18810167835

ChanganVR, 2020-06-09:
I don't use WeChat that much. If you have questions, I'd recommend you to leave your comments or questions here.

asenasen123, 2020-06-17:
你好，不好意思打扰了！因为运行cadrl(https://github.com/ChanganVR/CADRL.git)有问题，所以后来我在CrowdNav(https://github.com/vita-epfl/CrowdNav.git)运行cadrl程序。但我发现这两个程序中train_episode不一样（cadrl是1000，CrowdNav=10000）。我想请问下，如果单纯想测试cadrl，可以设置train_episode=1000,会收敛吗？

ChanganVR, 2020-06-17:
Hi, I don't remember the training parameters for them exactly, but you should definitely try the default first and based on the training/val curve, you can adjust the learning schedule accordingly.