vita-epfl/CrowdNav #17: How to set self.kinematics while running CADRL

Shuijing725, 2019-10-26:
In the CADRL paper by Chen et al, the heading angle of the robot (i.e. self.robot.theta) is part of the robot's state. But the default kinematic of this repo, as defined in crowd_anv/configs/policy.config, is "holonomic", and when kinematic is set to "holonomic", the angle theta never changes throughout an episode (it is always set to 90 degrees). 

So should I set kinematic to "unicycle" while training or testing CADRL?

Thank you a lot!

ChanganVR, 2019-11-07:
Hi, if I remember correctly, we tested this feature of unicycle kinematics after the submission and it worked. But since we didn't report the results of the unicycle constraint at the time of submission, the default value of kinematics was set to holonomic. 

Please go ahead and use this feature.

closed 2019-11-19 by ChanganVR

asenasen123, 2020-06-04:
> In the CADRL paper by Chen et al, the heading angle of the robot (i.e. self.robot.theta) is part of the robot's state. But the default kinematic of this repo, as defined in [...]

你好！我也是这个方向的研究生，有些问题想要请教，能否加下qq:1414396376,谢谢了