vita-epfl/CrowdNav #20: Visible

tessavdheiden, 2019-12-05:
Hi Changan,

You mentioned that setting the robot to visible lets the robot learn that humans avoid it.

The videos on the readme, are they in the unvisible setting?

ChanganVR, 2019-12-05:
Yes. And all the results in my paper are evaluated in the invisible setting.

anlanxuan, 2019-12-14:
Hi Changan,  I have no permission to download your data set and verification set in China. I hope you can provide me with some data.Can you send some data to me?Thanks

ChanganVR, 2019-12-16:
@anlanxuan There is no data to download except the repo itself. Validation and test set are just different random seeds.

closed 2019-12-16 by ChanganVR

anlanxuan, 2019-12-17:
Thanks, I mean what is the format of your ORCA data set, is there only location points? Can you give me the format of the input data?

ChanganVR, 2019-12-17:
@anlanxuan that's what I meant. There is not really a dataset in the repo. All the trajectories or seeds are hardcoded in the repo/simulation file. I recommend you to take a loot at this file: [](https://github.com/vita-epfl/CrowdNav/blob/master/crowd_sim/envs/crowd_sim.py)