vita-epfl/CrowdNav #21: Grid locations to be corrected?

tessavdheiden, 2020-03-17:
Hi Changan!

Super code! In multi_human_rl.py you compute the locations incorrectly. Please alter this piece of code:
line 154:

`                           dm[int(index)].append(1)
                            dm[int(index) + self.cell_num ** 2].append(other_vx[i])
                            dm[int(index) + self.cell_num ** 2 * 2].append(other_vy[i])
`

ChanganVR, 2020-04-01:
Hi Tessa,

I think the code is correct. The range of index goes from 0 to self.cell_num ** 2 - 1. It is the index of cells on the map. Why do you think it's incorrect? 

Changan

closed 2020-04-16 by ChanganVR

tessavdheiden, 2020-04-17:
Hi, 

First you make a list (0 to cell_num ** 2 * om_channel_size):
`dm = [list() for _ in range(self.cell_num ** 2 * self.om_channel_size)]`
[here](https://github.com/vita-epfl/CrowdNav/blob/64f4f412245ddb4de138de5708075d97d02b5034/crowd_nav/policy/multi_human_rl.py#L147) 
Then you fill in the cells:
`dm[3 * int(index) + 2].append(other_vy[i])`
[here](https://github.com/vita-epfl/CrowdNav/blob/64f4f412245ddb4de138de5708075d97d02b5034/crowd_nav/policy/multi_human_rl.py#L156)
the maximum index =  cell_num ** 2 + cell_num which is lower than the size of the list.

tessavdheiden, 2020-04-17:
Hi Changan,

You are right, I thought you wanted to do something in the image (lower part), but yours (upper part) is also correct!
[image]