vita-epfl/CrowdNav #26: Some weight might be missing

jiangxkjohn, 2020-06-16:
Hi, 

I try to run `python test.py --policy sarl --model_dir data/output --phase test --visualize --test_case 0` in README.md after the training step is over. But some error is happened like this:

```
2020-06-16 11:53:35, INFO: Using device: cpu
2020-06-16 11:53:35, INFO: Policy: SARL w/ global state
Traceback (most recent call last):
  File "test.py", line 113, in <module>
    main()
  File "test.py", line 59, in main
    policy.get_model().load_state_dict(torch.load(model_weights))
  File "/home/jxk/anaconda3/lib/python3.5/site-packages/torch/nn/modules/module.py", line 769, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for ValueNetwork:
	Missing key(s) in state_dict: "mlp1.0.weight", "mlp1.0.bias", "mlp1.2.weight", "mlp1.2.bias", "mlp2.0.weight", "mlp2.0.bias", "mlp2.2.weight", "mlp2.2.bias", "attention.0.weight", "attention.0.bias", "attention.2.weight", "attention.2.bias", "attention.4.weight", "attention.4.bias", "mlp3.0.weight", "mlp3.0.bias", "mlp3.2.weight", "mlp3.2.bias", "mlp3.4.weight", "mlp3.4.bias", "mlp3.6.weight", "mlp3.6.bias". 
	Unexpected key(s) in state_dict: "value_network.0.weight", "value_network.0.bias", "value_network.2.weight", "value_network.2.bias", "value_network.4.weight", "value_network.4.bias", "value_network.6.weight", "value_network.6.bias".
```
I run the process in README.md step by step, Can you tell what is going on or something I miss? 

Thank you!

ChanganVR, 2020-06-16:
It seems like the trained model weights is not matching the required model weights, which shouldn't be the case if you run the same code. Did you make any modifications to the code between training and testing?

A simple fix would be changing the name of weights from "value_network" to "mlp". But there is probably something else wrong with your training. You can print your model weight names out during training to see if they match the saved model weights.

jiangxkjohn, 2020-06-16:
I didn't modified the code except

https://github.com/vita-epfl/CrowdNav/blob/64f4f412245ddb4de138de5708075d97d02b5034/crowd_nav/train.py#L57#58

I find it occur a error and I commented them out. I think this two lines cannot modify the basic net, so it make me confused about it.

jiangxkjohn, 2020-06-16:
Sorry, I find that my train is no parameter and so that the default set is cadrl. but when I try to run

`python test.py --policy cadrl --model_dir data/output --phase test --visualize --test_case 0`

to see what will be going on in the model of cardrl. but It occur that this error:

```
2020-06-16 15:47:35, INFO: Using device: cpu
2020-06-16 15:47:35, INFO: Policy: CADRL without occupancy map
2020-06-16 15:47:35, INFO: human number: 5
2020-06-16 15:47:35, INFO: Not randomize human's radius and preferred speed
2020-06-16 15:47:35, INFO: Training simulation: circle_crossing, test simulation: circle_crossing
2020-06-16 15:47:35, INFO: Square width: 10.0, circle width: 4.0
2020-06-16 15:47:35, INFO: Agent is invisible and has holonomic kinematic constraint
Segmentation fault (core dumped)
```

I wonder whether you met it before? Thank you!

jiangxkjohn, 2020-06-16:
I try to run `test.py` by using GPU, and finally it works well.

ChanganVR, 2020-06-17:
Hi John, good to know the code is working, although I don't think it has anything to do with GPU/CPU training. I'm closing this issue.

closed 2020-06-17 by ChanganVR