vita-epfl/CrowdNav #30: some questions  about setup

hurong971013, 2020-10-27:
hi,sorry to bother  you .i can not understand the command in the setup,which means"install  crowd_nav into pip",as i know,the crowd_nav is a folder name,  how should i do to install this requirments. i  would like to seek for your help , thank you very much .

ChanganVR, 2020-10-31:
Hi, you simply need to run the command below that sentence "pip install -e .". What it does is to install this code repository as module into pip in development mode.  Let me know if you run into any problems after running that command.

hurong971013, 2020-11-03:
Thank you for your reply.when I try the command "pip install -e .". There is an error"FileNotFoundError: [Errno 2] 没有那个文件或目录: '/home/hurong/.virtualenvs/CrowdNav/lib/python3.5/site-packages/crowdnav-0.0.1-py3.5.egg'"
 I don't know where the problem is.Could you please tell me?

hurong971013, 2020-11-03:
I don’t know if I installed the Python-RVO2 library correctly. First of all, I downloaded the code of the Python-RVO2 library and installed it in this folder. I did not put the code folder of the Python-RVO2 library under the CrowdNav folder. Is this correct?
then，I enter the Python-RVO2-master folder and run these command："pip install -r requirements.txt" “python setup.py build”
"python setup.py install" 
At last it appeared this prompt：“Installed /home/hurong/.virtualenvs/CrowdNav/lib/python3.5/site-packages/pyrvo2-0.0.0-py3.5-linux-x86_64.egg
Processing dependencies for pyrvo2==0.0.0
Finished processing dependencies for pyrvo2==0.0.0
”，This should mean that I installed the Python-RVO2 library successfully, right?

ChanganVR, 2020-11-04:
> Thank you for your reply.when I try the command "pip install -e .". There is an error"FileNotFoundError: [Errno 2] 没有那个文件或目录: '/home/hurong/.virtualenvs/CrowdNav/lib/python3.5/site-packages/crowdnav-0.0.1-py3.5.egg'"
> I don't know where the problem is.Could you please tell me?

Where did you run this command? Did you run it in the root directory? The error basically says this repo is not correctly installed in the pip.

ChanganVR, 2020-11-04:
> I don’t know if I installed the Python-RVO2 library correctly. First of all, I downloaded the code of the Python-RVO2 library and installed it in this folder. I did not put the code folder of the [...]

I think so. Can you run the training script successfully?

hurong971013, 2020-11-05:
> Where did you run this command? Did you run it in the root directory? The error basically says this repo is not correctly installed in the pip.

I  download your code in the root directory.  The folder name is CrowdNav-master. And  I enter this folder,run this command "pip install -e .".Today I try this command again.  It shows"Installing collected packages: crowdnav
  Attempting uninstall: crowdnav
    Found existing installation: crowdnav 0.0.1
    Uninstalling crowdnav-0.0.1:
      Successfully uninstalled crowdnav-0.0.1
  Running setup.py develop for crowdnav
Successfully installed crowdnav
"
Is that right? But it does not mention the crowd_sim.

hurong971013, 2020-11-05:
> I think so. Can you run the training script successfully?

no,I can not. when I run the command "python train.py --policy sarl",it shows:
Traceback (most recent call last):
  File "train.py", line 177, in <module>
    main()
  File "train.py", line 57, in main
    repo = git.Repo(search_parent_directories=True)
  File "/home/hurong/.virtualenvs/CrowdNav/lib/python3.5/site-packages/git/repo/base.py", line 181, in __init__
    raise InvalidGitRepositoryError(epath)
git.exc.InvalidGitRepositoryError: /home/hurong/CrowdNav-master/crowd_nav

And I think the problem maybe is I did not install crowd_nav and crowd_sim.I really do not know how to fix it.

ChanganVR, 2020-11-05:
This is not necessarily because of incorrect installation of the code. This block of code is searching for .git folder in the directory. Make sure you have this folder in the parent directory.

If there is .git folder and this erro still occurs, you may comment out these few lines and see if training can run properly.

hurong971013, 2020-11-06:
> 

I do not have this .git folder in the CrowdNav directory.Is my installation of the code wrong or I lose the floder?

ChanganVR, 2020-11-06:
You probably downlowded the zip file. Just remove your current folder and run the code below again
`git clone git@github.com:vita-epfl/CrowdNav.git`

shuyaosong, 2020-11-09:
Dear Mr.Chen, 
Thanks for your advise, after remove my zip file and run 'git clone https://github.com/vita-epfl/CrowdNav.git', ours team solve this problem,thank you for your patience.

Regards.