vita-epfl/CrowdNav #35: About using the GPU

luxiaojuntokyo, 2021-01-24:
When I run: python train.py --policy sarl
Every time there will be: INFO: Using device: cpu

I wonder how can I use GPU to speed up the training?

Thanks a lot.

YuejiangLIU, 2021-01-26:
Please try and add the ['--gpu' argument](https://github.com/vita-epfl/CrowdNav/blob/6c4493201e9d892479dfae2ab378bdbec327314f/crowd_nav/train.py#L26) in your command

closed 2021-01-26 by YuejiangLIU

ChanganVR, 2021-01-26:
@luxiaojuntokyo the code runs faster on CPU because the speed bottleneck is on CPU-intensive operations. Feel free to explore and optimize the code such that you could use GPU to speed it up.

luxiaojuntokyo, 2021-01-28:
> @luxiaojuntokyo the code runs faster on CPU because the speed bottleneck is on CPU-intensive operations. Feel free to explore and optimize the code such that you could use GPU to speed it up.

Changan, thank you for your reply. If I get any result, I will comment here.

luxiaojuntokyo, 2021-01-28:
> Please try and add the ['--gpu' argument](https://github.com/vita-epfl/CrowdNav/blob/6c4493201e9d892479dfae2ab378bdbec327314f/crowd_nav/train.py#L26) in your command

LIU Yuejiang, thank you for your reply.