vita-epfl/CrowdNav #37: training issues

hurong971013, 2021-02-05:
Sorry to bother you, I want to do a comparative experiment of sarl. So the first step is that I need to train other algorithm， when I run the command "python train.py --policy orca", it prints:

(CrowdNav) hurong@hurong:~/CrowdNav/crowd_nav$ python train.py --policy orca
Output directory already exists! Overwrite the folder? (y/n)y
2021-02-05 12:56:38, INFO: Current git head hash code: %s
2021-02-05 12:56:38, INFO: Using device: cpu
usage: Parse configuration file [-h] [--env_config ENV_CONFIG]
                                [--policy POLICY]
                                [--policy_config POLICY_CONFIG]
                                [--train_config TRAIN_CONFIG]
                                [--output_dir OUTPUT_DIR] [--weights WEIGHTS]
                                [--resume] [--gpu] [--debug]
Parse configuration file: error: Policy has to be trainable

Could you please tell me why I can not train the orca?

ChanganVR, 2021-02-05:
Hi @hurong971013, ORCA is not parametric and does not require training. You should be able to evaluate this policy directly.

hurong971013, 2021-02-05:
> ORCA is not parametric and does not require training. You should be able to evaluate this policy directly.

Oh, that's it. Thank you. So when I run the command"python test.py --policy orca --phase test" , it prints:

2021-02-05 12:59:34, INFO: Using device: cpu
2021-02-05 12:59:34, INFO: human number: 5
2021-02-05 12:59:34, INFO: Not randomize human's radius and preferred speed
2021-02-05 12:59:34, INFO: Training simulation: circle_crossing, test simulation: circle_crossing
2021-02-05 12:59:34, INFO: Square width: 10.0, circle width: 4.0
2021-02-05 12:59:34, INFO: ORCA agent buffer: 0.000000
2021-02-05 12:59:34, INFO: Agent is invisible and has holonomic kinematic constraint
2021-02-05 12:59:42, INFO: TEST  has success rate: 0.43, collision rate: 0.57, nav time: 10.86, total reward: 0.0463
2021-02-05 12:59:42, INFO: Frequency of being in danger: 4.79 and average min separate distance in danger: 0.08
2021-02-05 12:59:42, INFO: Collision cases: 0 1 2 5 8 10 11 13 14 16 18 19 20 22 23 24 26 30 34 37 41 42 43 45 51 53 57 59 66 69 70 72 73 80 82 83 85 89 90 91 93 95 97 98 99 100 101 105 106 107 109 111 113 114 115 117 119 127 128 129 131 132 134 136 138 139 140 145 146 147 148 152 153 155 157 159 163 164 165 166 167 170 171 172 174 178 180 184 188 189 190 192 193 194 196 198 199 201 204 205 206 207 208 213 214 215 217 219 220 221 225 227 229 230 231 232 234 235 237 239 240 242 243 247 251 255 256 257 258 259 260 261 262 263 264 265 266 268 271 272 273 274 275 276 279 281 283 284 285 288 289 291 292 295 296 297 298 299 300 302 304 307 309 310 311 312 315 317 319 322 324 327 329 331 333 334 336 337 340 342 343 344 346 347 349 351 352 353 355 357 358 361 362 363 364 365 366 367 368 370 372 374 376 378 379 381 382 383 384 385 386 389 391 393 394 395 400 401 402 403 405 406 407 408 409 410 411 412 413 414 415 416 417 418 419 420 421 422 424 425 427 431 432 433 434 437 438 440 442 443 444 445 447 448 449 450 452 453 456 457 458 460 462 463 464 466 467 469 471 473 474 479 481 483 485 486 488 490 492 493 495 496 497 498
2021-02-05 12:59:42, INFO: Timeout cases: 118 168 224
 
what does these number means?

ChanganVR, 2021-02-06:
These numbers are the indices/ids of the failed cases. The success rate seems to be too low. Did you modify anthing in the code?

hurong971013, 2021-02-09:
> These numbers are the indices/ids of the failed cases. The success rate seems to be too low. Did you modify anthing in the code?

oh, I see. 
I haven't modified. Now just run all the comparison algorithms.

hurong971013, 2021-02-24:
> These numbers are the indices/ids of the failed cases. The success rate seems to be too low. Did you modify anthing in the code?

Hello, could you please tell me Is the robot in the code invisible by default?

ChanganVR, 2021-02-25:
Hi @hurong971013. yes, it is invisible by default.