vita-epfl/CrowdNav #38: drawing issue

hurong971013, 2021-03-04:
Sorry to bother you. Could you please teach me how  to draw a picture like this？（Trajectory comparison in an invisible test case in your paper's figure 5）
[image]

ChanganVR, 2021-03-07:
@hurong971013, the trajectory visualization code is here: https://github.com/vita-epfl/CrowdNav/blob/24be7ad39fed0658c5631c0820c67f9105e457bc/crowd_sim/envs/crowd_sim.py#L444-L484

and in the test.py, there are options to visualize the trajectory as well, which uses the code above.

705603621, 2021-05-21:
So how do we save this trajectory?I'm trying to type in python test.py --policy sarl --model_dir data/output --phase test --visualize --test_case 0 --traj. But I couldn't find the saved file.

ChanganVR, 2021-05-30:
This fucntion currently directly plots the figure in a pop-up window, you need to add a line of code to save the figure to the desired location.