vita-epfl/CrowdNav #39: Question about the formula calculating c value

guldamkwak3114, 2021-03-26:
In the <Crowd-Robot Interaction:
Crowd-aware Robot Navigation with Attention-based Deep Reinforcement Learning> paper,

[image]

what does this formula mean? softmax gives a vector and h_i is also a vector. what does mulitplication of softmax and h_I means?

ChanganVR, 2021-03-28:
hi @guldamkwak3114 we meant to normalize the scores alpha_i and do weighted sum of all h_i. Sorry for the bad and confusing writing.

guldamkwak3114, 2021-03-30:
What do you mean by normalize the scores alpha_i ?

ChanganVR, 2021-03-30:
Softmax is the normalization operation.

guldamkwak3114, 2021-03-30:
Still confuse me. softmax(alpha_i) and h_i are both vectors.................... how can we multiply them?

guldamkwak3114, 2021-03-30:
do you mean dot product of softmax(alpha_i) and h_i ?

ChanganVR, 2021-03-31:
The softmax(alpha_i) here means doing softmax for all scores and take the ith component, which is just a single value.  So we normalize the score with softmax for all neighbors and do a weighted sum of each neighbor's interaction feature.

guldamkwak3114, 2021-04-01:
alpha_i itself is a vector. then do you mean take softmax of (a_1,a_2, .... ,a_n) and get the i-th component? 

Thanks

ChanganVR, 2021-04-01:
Is alpha_i a vecotr? alpha_i is the score for each pair right? 

> take softmax of (a_1,a_2, .... ,a_n) and get the i-th component?
This is right.

guldamkwak3114, 2021-04-02:
according to this alpha_i should be length 100 vector. Am i miss-reading something? 
[image]

ChanganVR, 2021-04-02:
The [hidden units](https://stats.stackexchange.com/questions/181/how-to-choose-the-number-of-hidden-layers-and-nodes-in-a-feedforward-neural-netw) refers to all the MLP layers up to but not including the last layer. The last layer outputs one single value for each pair/human as the attention score. The corresponding code: https://github.com/vita-epfl/CrowdNav/blob/503173b836d5460e30234df7e14a7c67ee0ebfc7/crowd_nav/policy/sarl.py#L48