vita-epfl/CrowdNav #41: Action space question

hurong971013, 2021-08-09:
I'm sorry to bother you. Excuse me, in 80 discrete action spaces, why I'm sorry to bother you. Excuse me, in 80 discrete action spaces, why should the linear velocity be exponentially valued?

ChanganVR, 2021-08-11:
This is because we want the robot the take some more fine-grained actions when it gets close to the goal. Exponentially sampling leads to more low-valued velocities. We experimented with linear and expoential and found exponential worked better.

hurong971013, 2021-08-16:
> This is because we want the robot the take some more fine-grained actions when it gets close to the goal. Exponentially sampling leads to more low-valued velocities. We experimented with linear and expoential and found exponential worked better.

Thank you very much. I can understand.