vita-epfl/CrowdNav #42: compared algorithm question

hurong971013, 2021-08-09:
Could you please tell me the computional cost of ORCA, CADRL, LSTM-RL and SARL? 
I feel that these algorithms are not in the same category. It is not easy to compare. ORCA may perform best in other environments. Is that true? What is their computational cost?

ChanganVR, 2021-08-11:
What kinda of computational cost? Do you mean training or test time? ORCRA does not need training and the rest are learning based. At test time, I think ORCA also runs faster since the other are based on neural networks. But performance-wise, it is much lower than other algorithms.

hurong971013, 2021-08-16:
Thank you. I get it.

> What kinda of computational cost? Do you mean training or test time? ORCRA does not need training and the rest are learning based. At test time, I think ORCA also runs faster since the other are based on neural networks. But performance-wise, it is much lower than other algorithms.

Thank you. I get it.