vita-epfl/CrowdNav #47: Question in terms of the mlp1 and mlp2

chenzhutian, 2022-01-15:
Thanks for your great repo!
According to your paper, the `mlp1` is used to embed the joint state of robot and human, while the `mlp2` is used to obtain pairwise interaction features between robot and each human. However, based on the code, they look identical to me (both are a sequence of linear layers. perhaps the only differences is that mlp1 has a last relu but mlp2 doesn't). 

I am wondering why they (i.e., mlp1 and mp2) have different purpose if they are the same? Or why the outputs of `mlp1` are not the pairwise interaction features between robot and each human?

chenzhutian, 2022-01-16:
[image]
Or say, what are the differences between e_i and h_i? Why not use h_i for the attention, or e_i to calculate the crowd feature?

ChanganVR, 2022-01-18:
Hi @chenzhutian, my motivation behind this design was to allow the separation between interaction features learning and attention features learning. Despite the similarity of MLPs, `h_i` and `e_i` are followed by different networks and thus will receive different gradients in back propogation. Separate MLPs will allow these two vectors to learn and optimize for their desired functions in the whole pipeline.

chenzhutian, 2022-01-19:
Make sense, thanks for your explanation!

closed 2022-01-19 by chenzhutian