vita-epfl/CrowdNav #52: Number of humans in 'train' and 'test'

TUzizi, 2022-05-20:
Excuse me. 

I'm confused about the human_num in **crowd_sim.py**.
Why we should set human_num = 1 when self.robot.policy.multiagent_training = 1? 

`

    np.random.seed(counter_offset[phase] + self.case_counter[phase])
    if phase in ['train', 'val']:        
        human_num = self.human_num if self.robot.policy.multiagent_training else 1
        self.generate_random_human_position(human_num=human_num, rule=self.train_val_sim)
    else:
        self.generate_random_human_position(human_num=self.human_num, rule=self.test_sim)
    # case_counter is always between 0 and case_size[phase]
    self.case_counter[phase] = (self.case_counter[phase] + 1) % self.case_size[phase]
`
what I want to know is 
In training stage, do we use 1 human agent or 5 human agents (env.config [sim] human_num=5)?

I really need your help, thanks~

ChanganVR, 2022-05-20:
Multiagent-training differentiates CADRL from our policy. The former takes one human during training and ours takes 5 during training. At test time, both are tested on 5-human scenario.

TUzizi, 2022-05-20:
> Multiagent-training differentiates CADRL from our policy. The former takes one human during training and ours takes 5 during training. At test time, both are tested on 5-human scenario.

Oh, I see❤. Thanks.
But what makes the difference? I suppose that, In theory, cadrl could also have five people in training stage?

ChanganVR, 2022-05-20:
yeah possibly. I was just reproducing the training detailed in the paper

closed 2022-06-01 by TUzizi