vita-epfl/CrowdNav #55: I have two questions about testing

CAI23sbP, 2022-07-08:
Hi, i read your paper about CADRL. it is very awesome to me. 
By the way, i want to know there is possible about changing agent numbers.
(1) If i want to testing 6 human or robot, what can i do for code?
(2) After changing the code, training 6 agent environment, then would i be able to test that weight to 10 or more agent environment? CADRL , it would works well?

ChanganVR, 2022-07-08:
CADRL is not my work. It's a prior work that I implemented as a baseline. You can change the number of pedestrians easily whether during training or testing. As for if CADRL works well in more crowded scenarios, it's probably not as good as the other policies.

It does not have to the ORCA. If you can come with a human policy you think that's natural, you can just replace ORCA with that policy.

CAI23sbP, 2022-07-08:
> CADRL is not my work. It's a prior work that I implemented as a baseline. You can change the number of pedestrians easily whether during training or testing. As for if CADRL works well in more crowded scenarios, it's probably not as good as the other policies.
> 
> It does not have to the ORCA. If you can come with a human policy you think that's natural, you can just replace ORCA with that policy. I used ORCA because it's easily usable.

Thank you for your answer. It is very informational to me

closed 2022-07-08 by CAI23sbP