vita-epfl/CrowdNav #60: The environment must specify an action space

MERONAL, 2022-11-11:
Hi, I've managed to set up all the stuff but got the error message as below when I was running the train or test scripts:

```
Traceback (most recent call last):
  File "train.py", line 177, in <module>
    main()
  File "train.py", line 76, in main
    env = gym.make('CrowdSim-v0')
  File "/home/zy/miniconda3/envs/crowd_nav/lib/python3.6/site-packages/gym/envs/registration.py", line 669, in make
    env = PassiveEnvChecker(env)
  File "/home/zy/miniconda3/envs/crowd_nav/lib/python3.6/site-packages/gym/wrappers/env_checker.py", line 22, in __init__
    ), "The environment must specify an action space. https://www.gymlibrary.dev/content/environment_creation/"
AssertionError: The environment must specify an action space. https://www.gymlibrary.dev/content/environment_creation/

```

Any advice to fix this issue?

ChanganVR, 2022-11-14:
@MERONAL it looks like gym version is not compatible, have you tried to downgrade the gym version to an older version? You can check out previous issues, which mentioned similar problems.

MERONAL, 2022-11-16:
I reinstalled the environment and the issue is gone, not sure what happend thought. Thanks for your reply anyway.

closed 2022-11-16 by MERONAL