vita-epfl/CrowdNav #65: Can't run basic README tutorial

khalld, 2023-04-21:
Hi everyone, I'm not able to run the provided example given on README. I tested with three versions of python, respectively `3.6.15`, `3.7.13` , `3.11.1`. I'm not able to use `3.6.0` because is not supported for Mac OS 11+ (https://github.com/pyenv/pyenv/issues/2685#event-9052270090) . I receive an AssertionError:

After setup of rov2 and after `pip install -e` i try to run `python train.py --policy sarl`. For all version the error is the same:

```bash
/Users/danilo/GitHub/materiale-tesi/venv3615/lib/python3.6/site-packages/gym/core.py:27: UserWarning: WARN: Gym minimally supports python 3.6 as the python foundation not longer supports the version, please update your version to 3.7+
  "Gym minimally supports python 3.6 as the python foundation not longer supports the version, please update your version to 3.7+"
Output directory already exists! Overwrite the folder? (y/n)y
2023-04-21 16:47:38, INFO: Current git head hash code: %s
2023-04-21 16:47:38, INFO: Using device: cpu
2023-04-21 16:47:38, INFO: Policy: SARL w/ global state
Traceback (most recent call last):
  File "train.py", line 177, in <module>
    main()
  File "train.py", line 76, in main
    env = gym.make('CrowdSim-v0')
  File "/Users/danilo/GitHub/materiale-tesi/venv3615/lib/python3.6/site-packages/gym/envs/registration.py", line 669, in make
    env = PassiveEnvChecker(env)
  File "/Users/danilo/GitHub/materiale-tesi/venv3615/lib/python3.6/site-packages/gym/wrappers/env_checker.py", line 22, in __init__
    ), "The environment must specify an action space. https://www.gymlibrary.dev/content/environment_creation/"
AssertionError: The environment must specify an action space. https://www.gymlibrary.dev/content/environment_creation/
```

Anyone could help ?

khalld, 2023-04-23:
The above issue is caused by an upgraded version of `gym` library. Works correctly with version `0.18.0`, just downgrade it. Maybe could works also with `Python > 3.6.15`, anyone could confirm ?

closed 2023-04-23 by khalld

xiang2020git, 2023-05-12:
The sollution works in ubuntu 20.04 with python 3.8.10. Thanks!