vita-epfl/CrowdNav #66: Adding static obstacles

FADEAWAY114, 2023-04-27:
How about adding static obstacles to some locations in the environment?

ChanganVR, 2023-05-01:
This repo does not support adding static obstacles directly. But it shouldn't be too hard to do so and I've seen follow-up papers doing this. You can look up papers that cited this paper.

yuqikong, 2023-06-01:
Do you know which paper modified the environment and added static obstacles to the environment?

yuqikong, 2023-06-01:
> This repo does not support adding static obstacles directly. But it shouldn't be too hard to do so and I've seen follow-up papers doing this. You can look up papers that cited this paper.

Do you know which paper modified the environment and added static obstacles to the environment?

yuqikong, 2023-06-01:
> This repo does not support adding static obstacles directly. But it shouldn't be too hard to do so and I've seen follow-up papers doing this. You can look up papers that cited this paper.

May I ask if you know which paper modified the environment and added static obstacles to the environment, and it is best to add a laser radar.

JiayunjieJYJ, 2023-06-02:
> May I ask if you know which paper modified the environment and added static obstacles to the environment, and it is best to add a laser radar.

I have the same question

ChanganVR, 2023-06-05:
@yuqikong @JiayunjieJYJ a quick search on google scholar that cites this paper gives me this paper: "https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=9341540", I'm sure there are many others.

If you want to set human speed to zero to get a static obstacle, you can check the code here: https://github.com/vita-epfl/CrowdNav/blob/20d678085c06831e658a65b9e20c8bb6f6ecdc10/crowd_sim/envs/crowd_sim.py#L117C16-L142

closed 2023-06-05 by FADEAWAY114

JiayunjieJYJ, 2023-07-04:
> @yuqikong @JiayunjieJYJ a quick search on google scholar that cites this paper gives me this paper: "https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=9341540", I'm sure there are many [...]

Thanks! How to add static obstacles to the ORCA baseline in this code repo?

ChanganVR, 2023-07-06:
@JiayunjieJYJ it does not matter which policy you are using because you will be changing the simulation environment that equally applies to all policies, which also makes evaluation fair.

cocotorrow, 2024-02-06:
> How about adding static obstacles to some locations in the environment?
Pls have you solved this problem？

EVEREST-dlk, 2024-05-11:
> > This repo does not support adding static obstacles directly. But it shouldn't be too hard to do so and I've seen follow-up papers doing this. You can look up papers that cited this paper.
> 
> How to set a human's speed to zero to get a static obstacle?

Have you solved this? I also want to add static obstacles.