vita-epfl/CrowdNav #67: Settings for pedestrians in training and testing environments

yuqikong, 2023-05-26:
I am very interested in your work, I want to know how to set the position of pedestrians and the end point in your code during training and testing, and I noticed that the robot in your code is a holonomic robot, if I want to train a unicycle robot what should I do?

yuqikong, 2023-05-26:
I'd also like to know how to set the robot's start and end points, if possible.

ChanganVR, 2023-05-31:
@yuqikong please see this function for how we see the start and end position for humans: https://github.com/vita-epfl/CrowdNav/blob/20d678085c06831e658a65b9e20c8bb6f6ecdc10/crowd_sim/envs/crowd_sim.py#L155-L176. Robots always go upward.

For a unicycle robot, you can change the config here: https://github.com/vita-epfl/CrowdNav/blob/20d678085c06831e658a65b9e20c8bb6f6ecdc10/crowd_nav/configs/policy.config#L14. And here is how they are implemented: https://github.com/vita-epfl/CrowdNav/blob/20d678085c06831e658a65b9e20c8bb6f6ecdc10/crowd_sim/envs/crowd_sim.py#L336-L341.

yuqikong, 2023-06-01:
I got it, thank you!

closed 2023-06-01 by yuqikong