vita-epfl/CrowdNav #69: Process in trainning

JiangxinkeTao, 2023-10-25:
The following info is my problem, i hope that you can help me to solve the problem, thank you very much !!!

tomas@tomas:~/Documents/SocialNavi/CrowdNav/crowd_nav$ python3 train.py --policy sarl --gpu 
Output directory already exists! Overwrite the folder? (y/n)y
2023-10-25 23:09:22, INFO: Current git head hash code: %s
2023-10-25 23:09:22, INFO: Using device: cpu
2023-10-25 23:09:22, INFO: Policy: SARL w/ global state
2023-10-25 23:09:22, INFO: human number: 5
2023-10-25 23:09:22, INFO: Not randomize human's radius and preferred speed
2023-10-25 23:09:22, INFO: Training simulation: circle_crossing, test simulation: circle_crossing
2023-10-25 23:09:22, INFO: Square width: 10.0, circle width: 4.0
2023-10-25 23:09:22, INFO: Current learning rate: 0.010000
2023-10-25 23:13:50, INFO: TRAIN has success rate: 0.89, collision rate: 0.09, nav time: 12.18, total reward: 0.2416
Traceback (most recent call last):
  File "train.py", line 177, in <module>
    main()
  File "train.py", line 130, in main
    trainer.optimize_epoch(il_epochs)
  File "/home/tomas/Documents/SocialNavi/CrowdNav/crowd_nav/utils/trainer.py", line 39, in optimize_epoch
    outputs = self.model(inputs)
  File "/home/tomas/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/tomas/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/tomas/Documents/SocialNavi/CrowdNav/crowd_nav/policy/sarl.py", line 48, in forward
    scores = self.attention(attention_input).view(size[0], size[1], 1).squeeze(dim=2)
  File "/home/tomas/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/tomas/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/tomas/.local/lib/python3.8/site-packages/torch/nn/modules/container.py", line 215, in forward
    input = module(input)
  File "/home/tomas/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/tomas/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/tomas/.local/lib/python3.8/site-packages/torch/nn/modules/linear.py", line 114, in forward
    return F.linear(input, self.weight, self.bias)
RuntimeError: could not create a primitive descriptor for a matmul primitive

ChanganVR, 2023-11-01:
I don't think is an issue with this library. Looks like some package (version) issues. Does pytorch even work in your environment? If yes, you can start by removing codes and pinpointing what exactly is the issue. Have you tried the online solutions? e.g., https://github.com/pytorch/pytorch/issues/89104

JiangxinkeTao, 2023-11-02:
my previous python version is 3.8，i change it to 3.7，then the problem is solved, i forgot to close the issue , and thank you for your response sincerely.

closed 2023-11-02 by JiangxinkeTao